gdsfactory/gplugins #18: path_length_analysis notebook issue

joamatab, 2023-08-07:
the notebook has some issue 

https://gdsfactory.github.io/gplugins/notebooks/path_length_analysis.html

[image]

@tvt173

tvt173, 2023-08-08:
is this after a recent change? the image in the notebook on the docs looks good and correct

joamatab, 2023-08-08:
the image is a screenshot :) 

I don't know how to make bokeh display an image on the notebook

tvt173, 2023-08-08:
ah, yeah I understand that part. i was just saying that the screenshot version in the notebook looks correct. so it looks like it was working at some point, and i was curious if you knew when it broke

joamatab, 2023-08-08:
Im not sure but the tests were failing because gdsfactory/get_route was adding non serializable data, so i added clean_dict into it

joamatab, 2023-08-09:
it looks fixed now

closed 2023-08-09 by joamatab