gdsfactory/gplugins #104: femwell/solve_thermal.py

lukasc-ubc, 2023-09-09:
```
$ python solve_thermal.py 
Traceback (most recent call last):
  File "/Users/lukasc/Downloads/solve_thermal.py", line 219, in <module>
    from gplugins.gmsh.mesh2D import mesh2D
ModuleNotFoundError: No module named 'gplugins.gmsh.mesh2D'
```

- Mac OSX 11.6.7, Intel processor
- pip install: gdsfactory, gplugins, femwell

I have the latest version of gplugins (0.5.0)
```
$ pip install gplugins[gmsh,femwell]
Collecting gplugins
  Obtaining dependency information for gplugins from https://files.pythonhosted.org/packages/b9/dd/c790baea29ff148805030d562cbc63f6c0eaa4b428f7d672b5f8ddb7f847/gplugins-0.5.0-py3-none-any.whl.metadata
  Using cached gplugins-0.5.0-py3-none-any.whl.metadata (6.0 kB)
Requirement already satisfied: gdsfactory[cad]>=7.4.0 in /usr/local/lib/python3.10/site-packages (from gplugins) (7.4.6)
```

Seems like mesh2D is missing:

```
$ python 
Python 3.10.9 (main, Dec 15 2022, 18:20:40) [Clang 13.0.0 (clang-1300.0.29.30)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import gplugins.gmsh
>>> dir(gplugins.gmsh)
['MeshTracker', '__all__', '__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__path__', '__spec__', '__version__', 'annotations', 'break_geometry', 'cleanup_component', 'create_physical_mesh', 'fuse_polygons', 'get_layer_overlaps_z', 'get_layers_at_z', 'get_u_bounds_layers', 'get_u_bounds_polygons', 'get_uz_bounds_layers', 'list_unique_layerstack_z', 'map_unique_layerstack_z', 'mesh', 'mesh_from_polygons', 'meshtracker', 'order_layerstack', 'parse_component', 'parse_gds', 'parse_layerstack', 'refine', 'round_coordinates', 'tile_shapes', 'to_polygons', 'uz_xsection_mesh', 'xy_xsection_mesh']
```

Hoping that mesh2D was simply replaced by mesh, I made the change:

`    from gplugins.gmsh import mesh as mesh2D
`

And it runs, but then there is another error:

```
$ python solve_thermal.py 
dict_keys(['substrate', 'box', 'core', 'shallow_etch', 'deep_etch', 'clad', 'slab150', 'slab90', 'nitride', 'ge', 'undercut', 'via_contact', 'metal1', 'heater', 'via1', 'metal2', 'via2', 'metal3'])
2023-09-09 14:10:55.633 | WARNING  | gdsfactory.pdk:get_active_pdk:721 - No active PDK. Activating generic PDK.

2023-09-09 14:10:55.765 | INFO     | gdsfactory.technology.layer_views:__init__:790 - Importing LayerViews from YAML file: '/usr/local/lib/python3.10/site-packages/gdsfactory/generic_tech/layer_views.yaml'.
2023-09-09 14:10:55.766 | INFO     | gdsfactory.pdk:activate:334 - 'generic' PDK is now active
Traceback (most recent call last):
  File "/Users/lukasc/Downloads/solve_thermal.py", line 227, in <module>
    heater2 = gf.components.straight_heater_metal(length=50, heater_width=2).move(
  File "/usr/local/lib/python3.10/site-packages/gdsfactory/component.py", line 2153, in move
    raise ValueError(move_error_message)
ValueError: 
You cannot move a Component. You can create a new Component, add a reference to the other Component and then move the reference.

For example:

# BAD
c = gf.components.straight()
c.xmin = 10

# GOOD
c = gf.Component()
ref = c.add_ref(gf.components.straight()) # or ref = c << gf.components.straight()
ref.xmin = 10

```

joamatab, 2023-09-09:
Hi Lukas,

We are in the process of fixing the gdsfactory-femwell interface for thermal simulations, the idea is that from layout you will be able to directly extract the heater efficiency from the layout

in the meantime you will need to use femwell directly
https://helgegehring.github.io/femwell/photonics/examples/metal_heater_phase_shifter.html

@HelgeGehring 
@simbilod

HelgeGehring, 2023-09-09:
I think if you use 
```
c = gf.Component()
...
c.to_gmsh(
        type="uz",
       ...
)
```
it should work. I never used the mesh2D directly. @simbilod could you have a look as well?

simbilod, 2023-09-09:
Yes the 2D meshing interface is well-documented:

https://gdsfactory.github.io/gplugins/notebooks/meshing_03_2D_uz_mesh.html

Right now, to go beyond a mesh and do calculations requires a few manual steps, however. But in principle once you have the mesh as described above, you can use the femwell code directly from what Joaquin linked

joamatab, 2023-09-10:
Thank you for reporting this Lukas

Here is an initial fix
https://github.com/gdsfactory/gplugins/pull/106

We still need to add more useful examples on how to go from layout to thermal simulations

closed 2023-09-13 by joamatab