gdsfactory/gplugins #179: Include layer connectivity information in PDK

nikosavola, 2023-10-13:
**Is your feature request related to a problem? Please describe.**
Currently, [netlist extraction etc](https://gdsfactory.github.io/gplugins/notebooks/11_get_netlist_spice.html). with `get_l2n` does not take layer-to-layer connectivity into account unless manually given a klayout technology file. This should probably be included in a PDK and then the active one used be fetched

**Describe the solution you'd like**
Add layer connectivity information to PDK and use it in `get_l2n`.

joamatab, 2023-10-13:
You can get the technology object from the PDK after activating it

https://github.com/KLayout/klayout/issues/1223

nikosavola, 2023-10-16:
gdsfactory part for this was done in https://github.com/gdsfactory/gdsfactory/pull/2179

PR #185 (2023-10-17): Generate technology and support layer connectivity in `get_l2n`
This PR enables `get_l2n` to:
* generate a KLayout technology definition from the PDK on the get-go
* support connecting layers with vias as defined by KLayout Technologies. Implemented this way as recommended by https://github.com/KLayout/klayout/issues/1223

Closes #179
closed 2023-10-17 by joamatab