gdsfactory/gplugins #231: Schematic editor not opening

nikosavola, 2023-11-15:
**Describe the bug**
Is the schematic editor working? I installed `pip install -e .[schematic]` and tried running `20_schematic_driven_layout` in a raw Jupyter Notebook but the `SchematicEditor.visualize` method does nothing but print these lines.
[image]

Same can be seen in the docs: https://gdsfactory.github.io/gplugins/notebooks/20_schematic_driven_layout.html

Is there some bokeh configuration that needs to be done but is not documented?

joamatab, 2023-11-15:
@tvt173

joamatab, 2023-11-16:
it opens in Jupyter-lab but not in VSCode

[image]

nikosavola, 2023-11-16:
> it opens in Jupyter-lab but not in VSCode
> 
> [image]

I actually tried in jupyter lab, I'll also test now after the fix

nikosavola, 2023-11-16:
Doesn't work for me in Jupyter Lab still

joamatab, 2023-11-17:
The docs would not work without server 

Vscode was working well for me