gdsfactory/gplugins #262: Tidy3DComponent AttributeError: 'ComponentModeler' object has no attribute '_hash_self'

pascualmunoz, 2023-12-03:
**Describe the bug**
gplugins / tidy3d returns Tidy3DComponent AttributeError: 'ComponentModeler' object has no attribute '_hash_self'
**To Reproduce**
```python
from collections.abc import Callable
from functools import partial

from pydantic import BaseModel

import gdsfactory as gf
from gdsfactory.add_pins import add_pin_rectangle_inside
from gdsfactory.component import Component
from gdsfactory.config import CONF
from gdsfactory.cross_section import cross_section
from gdsfactory.technology import (
    LayerLevel,
    LayerStack,
    LayerView,
    LayerViews,
    LayerMap,
)
from gdsfactory.typings import Layer
from gdsfactory.config import print_version_pdks, print_version_plugins
from gdsfactory.generic_tech import get_generic_pdk

gf.config.rich_output()
nm = 1e-3

import matplotlib.pyplot as plt
import numpy as np
import tidy3d as td

import gplugins.tidy3d as gt
from gplugins import plot
from gplugins.common.config import PATH
import gplugins as gp

mat_Si = td.material_library["cSi"]["Li1993_293K"] # my_fitter(wvl_v, mat.si(wvl_v))
mat_SiO2 = td.material_library["SiO2"]["Horiba"] # my_fitter(wvl_v, mat.sio2(wvl_v))

class LayerMapMY(LayerMap):
    WG: Layer = (1, 0)

LAYER = LayerMapMY()

class MYLayerViews(LayerViews):
    WG: LayerView = LayerView(color="gold")

LAYER_VIEWS = MYLayerViews(layer_map=dict(LAYER))

def get_layer_stack_my(
        thickness_wg: float = 220 * nm,
) -> LayerStack:
    return LayerStack(
        layers=dict(
            strip=LayerLevel(
                layer=LAYER.WG,
                thickness=thickness_wg,
                zmin=0.0,
                material="Si",
            ),
        )
    )

LAYER_STACK = get_layer_stack_my()

WIDTH = 0.5

# Specify a cross_section to use
cs_strip = partial(gf.cross_section.cross_section, width=WIDTH, layer=LAYER.WG)

mmi1x2 = partial(
    gf.components.mmi1x2,
    width=WIDTH,
    width_taper=WIDTH,
    width_mmi=3 * WIDTH,
    cross_section=cs_strip,
)

generic_pdk = get_generic_pdk()

my_pdk = gf.Pdk(
     name="MY_PDK",
     cells=dict(mmi1x2=mmi1x2),
     cross_sections=dict(strip=cs_strip),
     layers=dict(LAYER),
     base_pdk=generic_pdk,
     layer_views=LAYER_VIEWS,
     layer_stack=LAYER_STACK,
)

my_pdk.activate()

sw_test = gf.components.straight(
    length = 10.0,
    width= 0.5,
    cross_section="strip",
)

sw_test.plot()

md = {
    "Si": mat_Si,
    "sio2": mat_SiO2,
}

sp = gt.write_sparameters(sw_test, plot_simulation_layer_name="strip", layer_stack=LAYER_STACK, material_mapping=md)
``` 

**Expected behavior**
gt.write_sparameters shows the cross-section to be simulated

**Suggested fix**
Sorry, not enough experience to answer this.

joamatab, 2023-12-03:
Hi Pascual,

yes, you need to update the tidy3d plugin specifically for that to work correctly,

can you try `pip install gplugins[tidy3d] --upgrade`?

Then, when running 
```
from gdsfactory.config import print_version_plugins

print_version_plugins()
```

you should use the latest tidy3d 2.5.0rc3, which is the pre-release for 2.5

@yaugenst 
@tyler-flex

joamatab, 2023-12-11:
we also released a new gplugins version 0.9.8 with the fix

so you could `pip install gplugins --upgrade`

pascualmunoz, 2023-12-11:
Hi, @joamatab  I can confirm it now runs smoothly.