gdsfactory/gplugins #291: sax mzi response looks like zero

joamatab, 2024-01-08:
After the latest gdsfactory / gplugins update tests are passing but now the docs look like they have an issue on the mzi response

https://gdsfactory.github.io/gplugins/notebooks/sax_01_sax.html

[image]

@flaport
@tvt173

flaport, 2024-01-08:
this is pretty weird, especially because the MC result seems to yield a nonzero result. I suspect that one of the default parameters for the nominal case is None, yielding nans (?) or something like that in the nominal case. Best to look at the parameters given to monte carlo and try to supply the same parameters with their nominal value to the nominal case. I can have a deeper look somewhere later this week.

flaport, 2024-01-15:
@joamatab , not sure what changes were made on the GDSFactory side or which decisions have been made, but this is a typical component dictionary of the exported netlist from GDSFactory:

```python
 'instances': {
        'bend_euler_1': {
            'component': 'bend_euler',
            'info': {
                'length': 16.637,
                'dy': 10.0,
                'radius_min': 7.061,
                'radius': 10.0,
                'width': 0.5,
                'route_info_type': 'xs_sc',
                'route_info_length': 16.637,
                'route_info_weight': 16.637,
                'route_info_xs_sc_length': 16.637,
                'route_info_n_bend_90': 1.0
            },
            'settings': {
                'radius': None,
                'angle': 90.0,
                'p': 0.5,
                'with_arc_floorplan': True,
                'npoints': None,
                'layer': None,
                'width': None,
                'direction': 'ccw',
                'cross_section': 'xs_sc',
                'add_pins': True,
                'add_bbox': None
            }
        },
```

clearly the settings dictionary is incomplete. Everything in `info` should also be in settings.

I also saw that the `full_settings` key during netlist export was removed. Why?

joamatab, 2024-01-16:
Yes, we have Component.info and Component.settings

We are getting ready for the kfactory move

It should be quite simple to merge both dictionaries

joamatab, 2024-01-18:
fixed by https://github.com/gdsfactory/gdsfactory/pull/2476

closed 2024-01-18 by joamatab