gdsfactory/gplugins #292: Add lumerical CHARGE integration

elamdf, 2024-01-08:
**Is your feature request related to a problem? Please describe.**
Being able to end-to-end simulate active devices designed in GDSFactory (such as PN phase shifters) in lumerical would be very attractive to those designing them. 

**Describe the solution you'd like**
What would you like to happen?
An integration with Lumerical CHARGE would allow users to specify the charge solver region, contacts, and bias sweep, copy over geometry + dopings from the GDSFactory design, and output a (3d for now, could add 2d functionality later) charge distribution that can be imported into FDTD.

elamdf, 2024-01-12:
seems like it won't be too different from FDTD- working on it over [here](https://github.com/elamdf/gplugins/tree/lumerical-charge). Would like to eventually have some easy way to get sparams for an active device under a sweep of voltage biases

simbilod, 2024-01-12:
Good luck! You might also want to look at photonflux: https://github.com/alexsludds/photonflux