gdsfactory/gplugins #328: Tidy3D WebError: The solver for v2.5.0rc3 is no longer supported

pascualmunoz, 2024-02-07:
**Describe the bug**
While submitting a component to Tidy3D using the gplugins interface, I get the error:

WebError: The solver for v2.5.0rc3 is no longer supported. If you still want to use this version, please contact our support team or upgrade your version to a newer version.

**To Reproduce**
```python
sp = gt.write_sparameters(
    component = mmi,
    layer_stack = LAYER_STACK,
    material_mapping = md, 
)
```

**Expected behavior**
Tidy3D runs simulation

**Suggested fix**
Unknown

**Versions**
What version of gplugins and each gplugin are you using? Notice that we only may be able help you if you are using the latest version.
You can find the version by running:

```python
Package ┃ version                                          ┃                                            Path ┃
┡━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┩
│     python │ 3.11.6 | packaged by conda-forge | (main, Oct  3 │             /Users/pascual/anaconda3/bin/python │
│            │ 2023, 10:37:07) [Clang 15.0.7 ]                  │                                                 │
│ gdsfactory │ 7.9.2                                            │ /Users/pascual/anaconda3/lib/python3.11/site-p… │
│   gplugins │ 0.9.8                                            │ ['/Users/pascual/anaconda3/lib/python3.11/site… │
│        ray │ 2.7.1                                            │ ['/Users/pascual/anaconda3/lib/python3.11/site… │
│    femwell │ 0.1.10                                           │ ['/Users/pascual/anaconda3/lib/python3.11/site… │
│     devsim │ 2.6.3                                            │ ['/Users/pascual/anaconda3/lib/python3.11/site… │
│     tidy3d │ 2.5.0rc3                                         │ ['/Users/pascual/anaconda3/lib/python3.11/site… │
│       meep │                                                  │                                                 │
│       meow │ 0.8.1                                            │ ['/Users/pascual/anaconda3/lib/python3.11/site… │
│     lumapi │ not installed                                    │                                                 │
│        sax │ 0.10.3                                           │ ['/Users/pascual/anaconda3/lib/python3.11/site…
```

joamatab, 2024-02-07:
Hi Pascual,

can you run `pip install gplugins[tidy3d] --upgrade`

our latest gplugins version is 0.9.12, which uses tidy3d==2.5.2, [See](https://github.com/gdsfactory/gplugins/blob/main/pyproject.toml)

pascualmunoz, 2024-02-07:
Up and runnning, thanks.

closed 2024-02-07 by pascualmunoz