gdsfactory/gplugins #344: process simulator

joamatab, 2024-02-16:
How could we make a 3D process simulator integration from layout so that we can simulate the geometry after fabrication?

at the moment we can only do 2D process simulations with [pyxs](https://github.com/gdsfactory/klayout_pyxs)

@tobre1
@nikosavola 
@simbilod 
@yaugenst 

https://github.com/ViennaTools/ViennaPS/issues/65

simbilod, 2024-02-16:
note that pyxs does not actually model fabrication in the way an actual process simulator would

From the creator: "Despite it's low visibility and obvious flaws, the project receives considerable attention by parts of the user community. Obviously, it is a great tool for teaching, layout comprehension and understanding device functionality and parasitic effects. But it may also be a waste of time. This is why I both love and hate it. "

https://wiki.f-si.org/index.php?title=KLayout_XSection_tool_-_Deep_insights_or_nonsense_in_colors%3F

simbilod, 2024-02-16:
it looks like they have a gds reader, which could be an easy starting point

https://github.com/ViennaTools/ViennaPS/blob/master/examples/GDSReader/GDSReader.py