gdsfactory/gplugins #361: Meshing errors when mesh with shapes with holes

Wangwei-Lan, 2024-03-18:
**Describe the bug**
When I want to mesh with shapes with holes, gdsfactory gives me an error "Exception: Curve loop is not closed".

**To Reproduce**
```python
import os
from math import inf
from pathlib import Path

import gdsfactory as gf
import pyvista as pv
from gdsfactory.components.interdigital_capacitor_enclosed import (
    interdigital_capacitor_enclosed,
)
from gdsfactory.generic_tech import LAYER, get_generic_pdk
from gdsfactory.technology import LayerStack
from gdsfactory.technology.layer_stack import LayerLevel
from IPython.display import display

from gplugins.palace import run_capacitive_simulation_palace
from gplugins.elmer import  run_capacitive_simulation_elmer

import gdsfactory as gf
import meshio
from gdsfactory.generic_tech import get_generic_pdk
from gdsfactory.pdk import get_layer_stack
from gdsfactory.technology import LayerStack
from skfem.io import from_meshio

from gplugins.gmsh.get_mesh import create_physical_mesh, get_mesh

gf.config.rich_output()
PDK = get_generic_pdk()
PDK.activate()

layer_stack = LayerStack(
    layers=dict(
        substrate=LayerLevel(
            layer=LAYER.WAFER,
            thickness=5,
            zmin=0,
            material="Si",
            mesh_order=99,
        ),
        bw=LayerLevel(
            layer=LAYER.WG,
            thickness=2,
            zmin=5,
            material="Nb",
            mesh_order=2,
        ),
    )
)
material_spec = {
    "Si": {"relative_permittivity": 11.45},
    "Nb": {"relative_permittivity": inf},
    "vacuum": {"relative_permittivity": 1},
}

simulation_box = [[-200, -200], [200, 200]]
c = gf.Component("simple_capacitance")

rect1 = gf.components.rectangle(size=[10,10],layer=(1,0),centered=True)
rect2 = gf.components.rectangle(size=[5,5],layer=(1,0),centered=True)
#
rect_with_holes = gf.Component("rect_with_holes")
rect_with_holes.add_ref(rect1)
rect_with_holes.add_ref(rect2)
rect_with_holes = gf.geometry.boolean(A=rect1,B=rect2,operation="not")
c.add_ref(rect_with_holes)

mesh = get_mesh(
    component=c,
    type="xy",
    z=5.05,
    layer_stack=layer_stack,
    filename="mesh.msh",
)
```

**Expected behavior**
```
Traceback (most recent call last):
  File "/Users/wangweilan/opt/anaconda3/envs/myaql/lib/python3.10/site-packages/IPython/core/interactiveshell.py", line 3577, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-2-0cc069127324>", line 1, in <module>
    runfile('/Users/wangweilan/Desktop/Work/ZaxisQuantum/projects/gdsfactory_tutorial/simple_capacitor.py', wdir='/Users/wangweilan/Desktop/Work/ZaxisQuantum/projects/gdsfactory_tutorial')
  File "/Users/wangweilan/Library/Application Support/JetBrains/Toolbox/apps/PyCharm-P/ch-0/231.9161.41/PyCharm.app/Contents/plugins/python/helpers/pydev/_pydev_bundle/pydev_umd.py", line 198, in runfile
    pydev_imports.execfile(filename, global_vars, local_vars)  # execute the script
  File "/Users/wangweilan/Library/Application Support/JetBrains/Toolbox/apps/PyCharm-P/ch-0/231.9161.41/PyCharm.app/Contents/plugins/python/helpers/pydev/_pydev_imps/_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "/Users/wangweilan/Desktop/Work/ZaxisQuantum/projects/gdsfactory_tutorial/simple_capacitor.py", line 74, in <module>
    mesh = get_mesh(
  File "/Users/wangweilan/opt/anaconda3/envs/myaql/lib/python3.10/site-packages/gplugins/gmsh/get_mesh.py", line 110, in get_mesh
    return xy_xsection_mesh(
  File "/Users/wangweilan/opt/anaconda3/envs/myaql/lib/python3.10/site-packages/gplugins/gmsh/xy_xsection_mesh.py", line 169, in xy_xsection_mesh
    return model.mesh(
  File "/Users/wangweilan/opt/anaconda3/envs/myaql/lib/python3.10/site-packages/meshwell/model.py", line 238, in mesh
    dimtags_out = entity_obj.instanciate()
  File "/Users/wangweilan/opt/anaconda3/envs/myaql/lib/python3.10/site-packages/meshwell/polysurface.py", line 105, in instanciate
    polysurface = self.get_gmsh_polygons()
  File "/Users/wangweilan/opt/anaconda3/envs/myaql/lib/python3.10/site-packages/meshwell/polysurface.py", line 67, in get_gmsh_polygons
    surfaces = [self.add_surface_with_holes(entry) for entry in self.polygons]
  File "/Users/wangweilan/opt/anaconda3/envs/myaql/lib/python3.10/site-packages/meshwell/polysurface.py", line 67, in <listcomp>
    surfaces = [self.add_surface_with_holes(entry) for entry in self.polygons]
  File "/Users/wangweilan/opt/anaconda3/envs/myaql/lib/python3.10/site-packages/meshwell/polysurface.py", line 87, in add_surface_with_holes
    exterior = self.model.add_surface(
  File "/Users/wangweilan/opt/anaconda3/envs/myaql/lib/python3.10/site-packages/meshwell/model.py", line 121, in add_surface
    channel_loop = self._channel_loop_from_vertices(vertices)
  File "/Users/wangweilan/opt/anaconda3/envs/myaql/lib/python3.10/site-packages/meshwell/model.py", line 111, in _channel_loop_from_vertices
    return self.occ.add_curve_loop(edges)
  File "/Users/wangweilan/opt/anaconda3/envs/myaql/lib/python3.10/site-packages/gmsh.py", line 7145, in addCurveLoop
    raise Exception(logger.getLastError())
Exception: Curve loop is not closed

```

**Versions**

The versions of python and gdsfactory, gplugins used are here: 

┃    Package ┃ version                        
│     python │ 3.10.6 (main, Oct 24 2022,    
│            │ 11:04:34) [Clang 12.0.0 ]     
│ gdsfactory │ 7.16.0                     
│   gplugins │ 0.9.13                       
│        ray │ not installed              
│    femwell │ 0.1.10                     
│     devsim │ 2.7.1                       
│     tidy3d │ 2.5.2                   
│       meep │ not installed               
│       meow │ 0.9.0                     
│     lumapi │ not installed          
│        sax │ 0.11.4

simbilod, 2024-03-18:
Yes we have seen this before, and had a local fix.

Pushed in #362 

Now your geometry meshes:

[image]

simbilod, 2024-03-18:
@Wangwei-Lan you might also want to 

(1) add an air/vacuum layer to your layerstack, 
(2) pad the device a bit in x-y

right now it's only the device and substrate

[image]

closed 2024-03-18 by simbilod

Wangwei-Lan, 2024-03-20:
@simbilod May I ask when will this local fix be available? I would like to try this out with Elmer and Palace. Thanks very much for the help.

Wangwei-Lan, 2024-03-20:
I managed to fix the bug by following #362, thanks very much.

simbilod, 2024-03-20:
> I managed to fix the bug by following #362, thanks very much.

Great! @joamatab will eventually merge the PR and make a new release to make this part of the pip package :)