gdsfactory/gplugins #390: Support schematic driven layout

joamatab, 2024-05-02:
Add examples and tests for schematic driven layout to support going from schematic to layout using different schematic capture tools

- [ ] from Ansys Lumerical Interconnect @kithminrw
- [ ] from Xschem @dquintero
- [ ] from Qucs @nikosavola 
- [ ] from [hdl21](https://github.com/dan-fritchman/Hdl21/blob/main/pdks/Sky130/sky130_hdl21/test_netlists.py) @dan-fritchman

daquintero, 2024-05-21:
Btw, related to https://github.com/gdsfactory/gdsfactory/issues/2755, on the hdl21 and gdsfactory integration:

Have been looking into it. Worth mentioning, currently the hdl21-sky130 package has the latest sky130nm PDK primitives. 

As per https://github.com/gdsfactory/skywater130/issues/63, we need to add the latest PDK layout instances into sky130nm in order to be able to link the layout and hdl21 schematic instances inherently. Currently we only have the digital cells in the sky130nm PDK, and I've tested this PR and it doesn't quite solve the problem as it depends on the klayout raw instantiation rather than one that is accessible inherently from gdsfactory https://github.com/gdsfactory/skywater130/pull/67. 

I will try importing the gds directly, but don't know the limitations of the layout parametrization after inputting the params from hdl21 which might be tricky.

dan-fritchman, 2024-05-24:
So @daquintero - correct me if I'm misunderstanding here:

- GdsFactory has a set of p-cell parameters per transistor, [e.g. these](https://github.com/mabrains/skywater130/blob/9d7ecf1deeadf09708172e7a0a3f0d3f4381f81f/sky130/cells/klayout/pymacros/cells/draw_fet.py#L936)
- Hdl21 has its own params for those transistors, [e.g. these](https://github.com/dan-fritchman/Hdl21/blob/cf3fc7b2c2b9542d9f16617d2d61c2af02439391/pdks/Sky130/sky130_hdl21/pdk_data.py#L52)
- The goal is to have one drive the other (or have some 3rd thing which drives both)

I understand that right?

daquintero, 2024-05-27:
Hi Dan! Yeah, I guess the idea is to ultimately pseudo-impelment the way the layout is generated from a schematic like in Virtuoso as you suggest. The method required would translate between the hdl21 params implementation and the instance/params schematic-editor implementation within gdsfactory. 

As far as I can tell so far there are a few steps to achieve this:
- [ ] Map sky130 spice model names to either gds instances/parametrized functions (such as between sky130_hdl21 and sky130 gdsfactory)
- [ ] Verify the correct mapping between params
- [ ] Test this out in the gplugins schematic editor.

A next stage could be finding a way to map between `xschem` and `hdl21/vlsir` protos. Maybe you already know if this is possible? Like creating a sky130nm schematic in xschem and exporting that to a vlsir representation/data that we could interact with in hdl21. I don't really know how that would work exactly as have been focused on the specific SPICE hdl21/vlsir-gdsfactory interface first. 

What do you think? How could something like this be better implemented?

daquintero, 2024-06-25:
So thanks Dan for the reference: 
- hdl21 have their own schematic editor implementation too https://www.generation-alpha-transistor.com/

I just recently bumped into this:
- https://kwantaekim.github.io/2024/06/18/Glayout/ which uses the gdsfactory pdks