gdsfactory/gplugins #406: kfactory-backend fails test

nikosavola, 2024-05-24:
**Describe the bug**
Newest gdsfactory major version with the kfactory-backend is failing the tests in CI. There seem to be couple of different type of errors.

[image]

See for example https://github.com/gdsfactory/gplugins/actions/runs/9203175080/job/25314251928

joamatab, 2024-06-05:
See WIP for the gdsactory release here
https://github.com/gdsfactory/gplugins/pull/413

It would be great if you could help fixing the plugins with the new gdsactory8 backend release, which we'll release in a day or so

- [ ] meshing
- [ ] tidy3d
- [ ] elmer, palace

@simbilod 
@HelgeGehring 
@yaugenst

simbilod, 2024-06-09:
meshing tests are now passing and meshes generated properly. See new issue to track further migration improvements

joamatab, 2024-06-10:
Awesome, thank you Simon!

joamatab, 2025-01-04:
I'll close this for now and make a new issue for fixing elmer

closed 2025-01-04 by joamatab