gdsfactory/gplugins #417: Finish porting meshing

simbilod, 2024-06-09:
Tests are now passing and meshes generated, but some things still need to be ported better

- [ ] don't hardcode the database unit when converting KLayout polygons to shapely for passing into meshwell
- [ ] consider doing the polygon preprocessing in KLayout instead of shapely (only convert to shapely when interfacing with meshwell)
- [ ] fix edge ports
- [ ] fix get_component_with_net_layers
- [ ] figure out issues with meshbool and physical overrides not being default=None
- [ ] fix warnings

simbilod, 2024-08-15:
get_component_with_net_layers fixed in #469 , thanks @mdecea

joamatab, 2024-10-17:
any updates on this Simon?

simbilod, 2024-10-17:
The code gives good outputs, fixing these is not high priority right now. Feel free to close. Happy to give pointers to whoever would want to pick this up.

closed 2025-01-05 by joamatab