gdsfactory/gplugins #437: finish porting plugins to gdsfactory8

joamatab, 2024-07-10:
Need to finish porting all plugins to gdsfactory8

- [ ] meep
- [ ] path_length_analysis

joamatab, 2024-07-17:
@nikosavola could you help with elmer and palace?

nikosavola, 2024-07-17:
> @nikosavola could you help with elmer and palace?

Sure, I haven't checked what fails yet.

nikosavola, 2024-08-08:
For capacitive simulations, this gmsh related function is not yet ported https://github.com/gdsfactory/gplugins/blob/60503bbb82aaae3d2146f69e0dff0399c1a14461/gplugins/common/utils/get_component_with_net_layers.py#L37-L61

joamatab, 2024-08-08:
@simbilod

simbilod, 2024-08-08:
> @simbilod

duplicate of #417

simbilod, 2024-08-15:
> For capacitive simulations, this gmsh related function is not yet ported
> 
> https://github.com/gdsfactory/gplugins/blob/60503bbb82aaae3d2146f69e0dff0399c1a14461/gplugins/common/utils/get_component_with_net_layers.py#L37-L61

presumably fixed in #469