gdsfactory/gplugins #452: `extend_ports` is defined but not actually used in `Tidy3DComponent`

tomflexcompute, 2024-08-01:
In [`Tidy3DComponent`](https://github.com/gdsfactory/gplugins/blob/94a3537655f8033333e845716a50c6d330c8b021/gplugins/tidy3d/component.py#L74), `extend_ports` is defined but not actually used anywhere. Therefore there is no way to effectively set the extension length for ports.

joamatab, 2024-08-03:
@yaugenst

yaugenst, 2024-08-03:
Hey @tomflexcompute, it's a field of the `LayeredComponentBase`, from which the `Tidy3DComponent` inherits, [see here](https://github.com/gdsfactory/gplugins/blob/main/gplugins/common/base_models/component.py). It is used to extend the ports of the gds component and set bounding boxes accordingly.

tomflexcompute, 2024-08-03:
> Hey @tomflexcompute, it's a field of the `LayeredComponentBase`, from which the `Tidy3DComponent` inherits, [see here](https://github.com/gdsfactory/gplugins/blob/f9e7b537fb9a85bc59ea7086e6af3da57afbaa45/gplugins/common/base_models/component.py#L58-L60). It is used to extend the ports of the gds component and set bounding boxes accordingly. Is there something in particular that you think isn't working?

Good to see you on the other side @yaugenst :grin:. Some user reported this parameter has no effect. Let me double check.

tomflexcompute, 2024-08-07:
Ok the user confirmed that it's working properly even though he encountered some issues that he couldn't reproduce now. If he encounters the issue again we can look into it.

closed 2024-08-07 by tomflexcompute

yaugenst, 2024-08-07:
Great!