gdsfactory/gplugins #477: gm.get_meep_adjoint_optimizer function

kdotom, 2024-08-23:
**Describe the bug**
`
/path/to/miniconda3/envs/mp/lib/python3.11/site-packages/gplugins/gmeep/meep_adjoint_optimization.py:148: UserWarning: Setting `rectangle_S5_5_LWG_CFal_d89a89d8_2500_2500.center` in um is deprecated and will change to DataBaseUnits in gdsfactory9. Please use `rectangle_S5_5_LWG_CFal_d89a89d8_2500_2500.dcenter` instead.
`

**To Reproduce**
```
import gdsfactory as gf
import gplugins as gp
import gplugins.gmeep as gm
import numpy as np
import meep as mp
from meep import MaterialGrid, Medium, Vector3, Volume
from meep.adjoint import (
    conic_filter,
    DesignRegion,
    get_conic_radius_from_eta_e,
    tanh_projection,
)

#### Demo to reproduce bug
Nx = 50
Ny = 50

c = gf.Component()
wg = c << gf.components.straight(layer=(1,0),width=0.5,length=10)
wg.dmovex(-7e3)

design_variables = mp.MaterialGrid(mp.Vector3(Nx, Ny), mp.Medium(index=1.444), mp.Medium(index=3.45), grid_type="U_MEAN")
design_region = \
    DesignRegion(
        design_variables,
        volume=Volume(
            center=Vector3(),
            size=Vector3(5, 5, 0),
        ),
    )

seed = 240
np.random.seed(seed)
x0 = np.random.rand(
        Nx * Ny,
    )

c.add_port("o1", port=wg.ports["o1"]) # assign ports to component
c.add_port("o2", port=wg.ports["o2"]) # assign ports to component

def J1(o1, o2):
    criterion = npa.abs(o2/o1)
    return criterion

gm.get_meep_adjoint_optimizer(
    c,
    J1,
    [design_region],
    [design_variables],
    x0,
    resolution=10,
    cell_size=(
        10,
        10,
    ),
    tpml=1.0,
    extend_ports_length=0,
    port_margin=1,
    port_source_name=f"o1",
    port_source_offset=-1.5,
    port_monitor_offset=-1.5,
    distance_source_to_monitors=1,
)
```

**Expected behavior**
Warning should not occur and design with a waveguide and design region (properly sized) should be produced.

**Suggested fix**
_Potentially_, Change block.center to block.dcenter on line 148 of path/to/site-packages/gplugins/gmeep/meep_adjoint_optimization.py
I haven't updated this as I'm not sure how changes will propagate through the gmeep library, given the new units update.

joamatab, 2024-10-17:
@simbilod 

if you have some time maybe you can take a look at the meep plugin

simbilod, 2024-10-17:
Hi @kdotom , I actually don't use MEEP anymore so if you have ideas on how to improve the plugin feel free to make any changes you see fit!