gdsfactory/gplugins #499: fix lumerical and meep plugins

joamatab, 2024-11-25:
Hi JD,

is this something you could take a look?

Yannick did a great job on the tidy3d plugin, so maybe we can do the other ones in a similar way

@jan-david-fischbach
@smartalecH
@simbilod

joamatab, 2025-03-27:
I think the new release 1.3.2 should fix the Lumerical plugin

@kdotom can you `pip install gplugins --upgrade` and test again?

kdotom, 2025-03-29:
Lumerical is now starting with the fix. Cladding and simulation region load. Python stops with this error:
`(1, 0) is not a <class 'gdsfactory.technology.layer_stack.LogicalLayer'>`
at this line:
`if layer not in component_layers:`

If we replace that with this:
`if not any([layer is element for element in component_layers]):`
we can use mixed types, but it skips over any layers that don't match exactly (i.e. tuples vs. layers). (Which makes sense)

We may need a means of translating tuples into layers for the comparison line above. I'll take a look to see how Yannick solved this with tidy3d.

joamatab, 2025-06-06:
both Meep and Lumerical are now fixed

fixed in https://github.com/gdsfactory/gplugins/pull/594

closed 2025-06-06 by joamatab