gdsfactory/gplugins #544: Cannot generate layout from schematic-driven layout module in docs

hezq12358, 2025-02-13:
**Describe the bug**
I have been trying to made a demo about schematic-driven layout based on the docs [gdsfactory.github.io/gplugins/notebooks/21_schematic_driven_layout_lumerical.html](url). However, I find that I can't generate a layout using the code listed in the docs. I have met some problems, which makes me guess that maybe I have done something incorrectly or there exists bugs in Schematic-dirven layout. These problems are:
1. Compared with original *.sp, there seems exists much information missing in the generated *.pic.yml, making it impossible to generate a valid layout.
2. I added some code, trying to generate a layout from generated *.pic.yml, but found the program fails with "Value error". 

The 3rd problem that I have is that I have read a closed issue [https://github.com/gdsfactory/gplugins/issues/231](url), which seemingly indicates that the gplugins in old version support schematic-driven layout. In that issue, there even exists a link [[gdsfactory.github.io/gplugins/notebooks/20_schematic_driven_layout.html](https://gdsfactory.github.io/gplugins/notebooks/20_schematic_driven_layout.html)](url) the old docs, in which introduces a more complete process of schematic-driven layout. However, I cannot access it now since it is invalid. I would appreciate it if anyone could provide any suggestions.

**To Reproduce**
I have changed the main function of a python file (<PKG_PATH>\gplugins\spice\tests\test_interconnect.py). And the changed main function is listed as below:
`if __name__ == "__main__":
    sche: str = "compound_1"
    netlist_path = GDSDIR_TEMP / "test_interconnect.sp"
    netlist_path.write_text(spice_netlist_interconnect)
    picyaml_path = GDSDIR_TEMP / f"{sche}.pic.yml"
    mapping_path = PATH.module / "lumerical" / "mapping_ubcpdk.yml"
    spice_to_yaml(
        netlist_path=netlist_path,
        picyaml_path=picyaml_path,
        mapping_path=mapping_path,
        pdk="ubcpdk",
    )
    s = yaml.safe_load(picyaml_path.read_text())
    # print(s)
    # print(len(s.instances))
    assert len(s["instances"]) == 1

    # region --- write_gds
    import gdsfactory as gf
    from gdsfactory import Component
    from matplotlib.pyplot import show

    GDS = GDSDIR_TEMP / "gds" / f"{sche}.gds"
    C: Component = gf.read.from_yaml(picyaml_path)
    C.write_gds(GDS)
    C.plot()
    show()
    # endregion`

other files and code are not changed.

**Expected behavior**
The generated *.pic.yml seemingly have insufficient information, which cannot be used to generate a valid layout.
`instances:
  X_COMPOUND_1:
    component: COMPOUND_1
    settings:
      Model_Version: '"2016/04/07"'
      coupling_length: '{{test_param1*1e-6}}'
      gap: 0.2
      library: '"design_kit/ebeam"'
      note: '".- The current model only supports "'
      radius: 5.0
      test_param1: 3.0
      test_param2: 2.0
      wg_length: 0.0001
      wg_width: 0.5
name: TOP
pdk: ubcpdk
placements:
  X_COMPOUND_1:
    mirror: false
    rotation: 0.0
    x: 75.0
    y: 0.0
routes: {}
`

The program fails with the following output:
`D:\core\anaconda4\envs\linksdk\python.exe D:\core\anaconda4\envs\linksdk\Lib\site-packages\gplugins\spice\tests\test_interconnect.py 
LOG: Non-PDK elements detected! Removing instances {X_dc_1, X_dc_2, X_dc_3, X_ebeam_y_1550_1, X_ebeam_y_1550_2} from netlist.
LOG: 0 electrical pads were added.
LOG: Non-PDK elements detected! Removing instances {(i.e.,, X_ebeam_y_1550_1, X_ebeam_y_1550_2, X_TE1550_SubGC_neg31_oxide_1, X_TE1550_SubGC_neg31_oxide_2, X_ebeam_y_1550_3, X_ebeam_y_1550_4, X_wg_heater_1, X_wg_heater_2, X_TE1550_SubGC_neg31_oxide_3, X_TE1550_SubGC_neg31_oxide_4} from netlist.
LOG: 0 electrical pads were added.
LOG: Overwrote C:\Users\hezhengqian\AppData\Local\Temp\gdsfactory\compound_1.pic.yml.
LOG: Overwrote C:\Users\hezhengqian\AppData\Local\Temp\gdsfactory\compound_1.pic.yml.
Traceback (most recent call last):
  File "D:\core\anaconda4\envs\linksdk\Lib\site-packages\gplugins\spice\tests\test_interconnect.py", line 93, in <module>
    C: Component = gf.read.from_yaml(picyaml_path)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\core\anaconda4\envs\linksdk\Lib\site-packages\gdsfactory\read\from_yaml.py", line 756, in from_yaml
    net = Netlist.model_validate(dct)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\core\anaconda4\envs\linksdk\Lib\site-packages\pydantic\main.py", line 596, in model_validate
    return cls.__pydantic_validator__.validate_python(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
pydantic_core._pydantic_core.ValidationError: 1 validation error for Netlist
instances.X_COMPOUND_1
  Value error, 'COMPOUND_1' not in PDK 'generic'. Did you mean ['C']? [type=value_error, input_value={'component': 'COMPOUND_1....0001, 'wg_width': 0.5}}, input_type=dict]
    For further information visit https://errors.pydantic.dev/2.9/v/value_error`

**Versions**
`                                    Modules                                    
┌────────────┬────────────────────────────────┬───────────────────────────────┐
│    Package │ version                        │                          Path │
├────────────┼────────────────────────────────┼───────────────────────────────┤
│     python │ 3.12.0 | packaged by Anaconda, │ D:\core\anaconda4\envs\links… │
│            │ Inc. | (main, Oct  2 2023,     │                               │
│            │ 17:20:38) [MSC v.1916 64 bit   │                               │
│            │ (AMD64)]                       │                               │
│ gdsfactory │ 8.18.2                         │ D:\core\anaconda4\envs\links… │
│   gplugins │ 1.2.4                          │ ['D:\\core\\anaconda4\\envs\… │
│        ray │ not installed                  │                               │
│    femwell │ not installed                  │                               │
│     devsim │ not installed                  │                               │
│     tidy3d │ not installed                  │                               │
│       meep │ not installed                  │                               │
│       meow │ not installed                  │                               │
│     lumapi │ not installed                  │                               │
│        sax │ 0.14.1                         │ ['D:\\core\\anaconda4\\envs\… │
└────────────┴────────────────────────────────┴───────────────────────────────┘`

joamatab, 2025-02-13:
Hi!  

The schematic-driven layout using Lumerical was a proof of concept that for me never worked.

I’d recommend considering [GDSFactory+](https://gdsfactory.com/) for a reliable schematic-driven layout flow.

hezq12358, 2025-02-19:
Hi!

I have watched some videos about gdsfactory+ and filled out a google form on the gdsfactory+ website. I think it's a powerful and easy-to-use tool for electronic-photonic chip development. However, several days later, I haven't receive any reply and I can't get access to this tool now.

What I want to do is to create a simple demo demonstrating the process of schematic-driven layout based on open-source tools. Could I use some functions of gdsfactory+ without subscription?

joamatab, 2025-04-12:
Hi,

Yes, you can do a fully open source schematic driven layout with Xschem, 

then use the spice to YAML converter to convert it to a GDSFactory netlist