gdsfactory/gplugins #575: gmeep.write_sparameters_meep not implementing geometry.

Arthaj-Octopus, 2025-04-23:
**Describe the bug**
gmeep.write_sparameters_meep not correctly implementing geometry.

**To Reproduce**
Installing gdsfactory and meep and running the tutorial at https://gdsfactory.github.io/gplugins/notebooks/meep_01_sparameters.html

**Expected behavior**
the waveguide to appear in the meep plot

**Suggested fix**
the command should work as intended.

**Versions**
                                                    Modules                                                      
┏━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┓
┃        Package ┃ version                                        ┃                                          Path ┃
┡━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┩
│         python │ 3.11.12 | packaged by conda-forge | (main, Apr │ /home/ammonite/anaconda3/envs/optics/bin/pyt… │
│                │ 10 2025, 22:23:25) [GCC 13.3.0]                │                                               │
│     gdsfactory │ 9.5.1                                          │ /home/ammonite/anaconda3/envs/optics/lib/pyt… │
│         devsim │ 2.9.1                                          │ ['/home/ammonite/anaconda3/envs/optics/lib/p… │
│        femwell │ 0.1.11                                         │ ['/home/ammonite/anaconda3/envs/optics/lib/p… │
│ gdsfactoryplus │ not installed                                  │                                               │
│       gplugins │ 1.3.3                                          │ ['/home/ammonite/anaconda3/envs/optics/lib/p… │
│       kfactory │ 1.4.4                                          │ ['/home/ammonite/anaconda3/envs/optics/lib/p… │
│         lumapi │ not installed                                  │                                               │
│           meep │ 1.30.0                                         │ ['/home/ammonite/anaconda3/envs/optics/lib/p… │
│           meow │ 0.13.0                                         │ ['/home/ammonite/anaconda3/envs/optics/lib/p… │
│            ray │ 2.44.1                                         │ ['/home/ammonite/anaconda3/envs/optics/lib/p… │
│            sax │ 0.14.5                                         │ ['/home/ammonite/anaconda3/envs/optics/lib/p… │
│         tidy3d │ 2.8.2                                          │ ['/home/ammonite/anaconda3/envs/optics/lib/p… │
│          vlsir │ not installed                                  │                                               │
└────────────────┴────────────────────────────────────────────────┴───────────────────────────────────────────────

[image]

[image]

[image]

[image]

[image]

joamatab, 2025-04-23:
@jan-david-fischbach
@simbilod 

is this something you could take a look at?

jan-david-fischbach, 2025-04-25:
Hey @joamatab, 
to me it seems like the culprit is in https://github.com/gdsfactory/gplugins/blob/73da7c653635cfe370619911ba18d41e74264c11/gplugins/gmeep/get_meep_geometry.py#L51
where layer_to_polygons is a dict with just the integer layer numbers as keys, while the layer stack contains logical and derived layers. I am not sure what would be the best way to make sure the layers are matched properly. Maybe @sebastian-goeldi has a good idea?

simbilod, 2025-04-25:
Layerlevel.layer.get_shapes will return the polygons after layer operations

threaddy, 2025-05-13:
I found a possible workaround by modifying the function **get_meep_geometry_from_component()**.
It seems to work with the tutorial notebook, but not sure if the results are correct. More testing needed. 
Here is the updated definition:

```
def get_meep_geometry_from_component(
    component: ComponentSpec,
    layer_stack: LayerStack | None = None,
    material_name_to_meep: dict[str, str | float] | None = None,
    wavelength: float = 1.55,
    is_3d: bool = False,
    dispersive: bool = False,
    **kwargs,
) -> list[mp.GeometricObject]:
    """Returns Meep geometry from a gdsfactory component.

    Args:
        component: gdsfactory component.
        layer_stack: for material layers.
        material_name_to_meep: maps layer_stack name to meep material name.
        wavelength: in um.
        is_3d: renders in 3D.
        dispersive: add dispersion.
        kwargs: settings.
    """
    component = gf.get_component(component=component, **kwargs)
    layer_stack = layer_stack or get_layer_stack()

    layer_to_thickness = layer_stack.get_layer_to_thickness()
    layer_to_material = layer_stack.get_layer_to_material()
    layer_to_zmin = layer_stack.get_layer_to_zmin()
    layer_to_sidewall_angle = layer_stack.get_layer_to_sidewall_angle()
    component_with_booleans = layer_stack.get_component_with_derived_layers(component)

    geometry = []
    layer_to_polygons = component_with_booleans.get_polygons_points()

    # ordered_layer_stack_keys = order_layer_stack(layer_stack)[::-1]

    for layer_tuple in component_with_booleans.layers:
        layer_index = get_layer(layer_tuple)
        layer = get_layer_name(layer_index)

        # layer_material 

        # if layer_index not in layer_to_polygons:
        #     continue
        polygons = layer_to_polygons[layer_index]

        print(f"layer: {layer}, polygons: {polygons}")

        if layer in layer_to_thickness and layer in layer_to_material:
            height = layer_to_thickness[layer] if is_3d else mp.inf
            zmin_um = layer_to_zmin[layer] if is_3d else 0
            # center = mp.Vector3(0, 0, (zmin_um + height) / 2)

            for polygon in polygons:
                vertices = [mp.Vector3(p[0], p[1], zmin_um) for p in polygon]
                material_name = layer_to_material[layer]

                if material_name:
                    material = get_material(
                        name=material_name,
                        dispersive=dispersive,
                        material_name_to_meep=material_name_to_meep,
                        wavelength=wavelength,
                    )
                    geometry.append(
                        mp.Prism(
                            vertices=vertices,
                            height=height,
                            sidewall_angle=np.pi * layer_to_sidewall_angle[layer] / 180
                            if is_3d
                            else 0,
                            material=material,
                            # center=center
                        )
                    )
    return geometry
```

[image]
[image]

closed 2025-05-25 by joamatab