gdsfactory/gplugins #616: Feature Request: Support for Sidewall Angles in Waveguide Structures in FDTD

pkqustc, 2025-07-12:
I would like to request the addition of sidewall angle support for waveguide structures in Lumerical FDTD. In many practical integrated photonics platforms—the waveguide sidewalls are not perfectly vertical and often exhibit a non-zero taper angle due to fabrication limitations (e.g., ICP-RIE processes).

joamatab, 2025-07-12:
I think it's already supported as the LayerStack sidewall_angle

pkqustc, 2025-07-12:
Hi @joamatab,

Thank you very much for the previous tip regarding LayerStack.sidewall_angle. It pointed me in the right direction for my investigation.

Following your suggestion, I've looked into this further, but I seem to have run into some confusion and was hoping you could help clarify.

In my local version of the code, I took a close look at the write_sparameters_lumerical function. Its core logic appears to only use a simple vertical extrusion method by calling gdsimport followed by setting the z span:

Python

# ... (inside write_sparameters_lumerical)
s.gdsimport(str(gdspath), "top", f"{layer_tuple[0]}:{layer_tuple[1]}")
layername = f"GDS_LAYER_{layer_tuple[0]}:{layer_tuple[1]}"
s.setnamed(layername, "z", z * 1e-6)
s.setnamed(layername, "z span", thickness * 1e-6)
# ...

It's entirely possible I'm looking in the wrong place or have missed a key part of the process. Any guidance would be greatly appreciated.

Thanks again for your patience and help!

closed 2025-07-22 by pkqustc